Darthpbal/eclipse-ASUPayload #11: luminosity

Darthpbal, 2017-05-17:


Darthpbal, 2017-05-17:
circuit configuration is very simple
sensor  => controller
VCC      => 3.3V
SDA      => SDA
SCL      => SCL
GND      => GND

Darthpbal, 2017-05-17:
interrupt feature offers a flag signalling when the next ADC value is available. This appears to provide no benefit since whenever the sensor is polled, it will provide the latest value anyway. 

If interrupt use is desired, connect a 10K -100K ohm pull up resistor is required as advised by adafruit (link)[https://learn.adafruit.com/tsl2561/wiring]

Darthpbal, 2017-05-17:
Board is 3.3V only, and uses a pull up resistor on the I2C network to become 5V compatible.

Darthpbal, 2017-05-17:
Now I remember the reason I was interested in the interrupt feature. 

If the integration time on the sensor is longer than desirable for the 15HZ log speed, as in the event that multiple logs happen while one light integration cycle occurs. We'd be able to cut poll time for this sensor but not even trying to poll it's value until the interrupt flag has been seen. Otherwise just log a stored previous value. 

It's not clear yet whether or not this will need to be implemented. But the faster controller clock speed will help find out.

bmpina, 2017-06-05:
Luminosity Sensor that was found in the lab is working great and can be used as a backup/redundancy

Darthpbal, 2017-06-15:
2 lum

bmpina, 2017-06-17:
ready for integration

bmpina, 2017-06-17:
adafruit breakout has less drift than the sparkfun one

bmpina, 2017-07-03:
Sparkfun UV board has the same sensor as the luminosity board. We most likely do not need this sensor and it will be used for backup as needed.