Darthpbal/eclipse-ASUPayload #12: UV

Darthpbal, 2017-05-17:


Darthpbal, 2017-05-19:
product [page](https://www.sparkfun.com/products/12705)

Darthpbal, 2017-05-19:
according to bianca's research:
uv sensors DO normally have a shelf life, and it's due to moisture exposure. 

UV sensor shelf lives abide by a standard name indicating how many days they will survive after opening up their packaging exposing them to moisture vs never opening them.

This makes determining the exact shelf life of our own UV sensors bought from those vendors since we can't really know if the UV breakout boards are fabricated in a cleanroom or otherwise dehumidified environment.

Darthpbal, 2017-05-19:
The sparkfun UV sensor we have in hand currently appears to be burnt. [Here](https://docs.google.com/spreadsheets/d/1r1zpF9WsUKP9OjmTeNd3Y-rwN-IBQu9Gf-9LlU6ysRk/edit) is the data from a test that give me that assumption. 

The test was done on a 9/10 UV index day at around 3pm with lots of sun out, and I walked between a dark hallway and into the sunlight with the sensor facing the sun. 

the results show the voltage output from the sensor only varies 0.1V of it's assumed ~2.5V output range.

Darthpbal, 2017-05-25:
**from biancas commit**

Confirmed that sensor is damaged. 

The ML8511 should not be outputting values over 15mW/cm^2

the sensor is consistently outputting values beyond this limit. I tried various hookup guides and the values are still out of range. 
Hookup guides I tried: 
https://learn.sparkfun.com/tutorials/ml8511-uv-sensor-hookup-guide
https://www.dfrobot.com/wiki/index.php/UV_Sensor_v1.0-ML8511_SKU:SEN0175
https://plot.ly/arduino/uv-sensor-tutorial/

Values should be close to zero indoors and around 5mW/cm^2 outdoors

This is a list of people who have tested with accurate results. 
My hookup and code was identical to theirs which leads me to believe the sensor is broken. 
http://rdiot.tistory.com/72
https://rydepier.wordpress.com/2016/03/31/arduino-and-ml8511-uv-detector/

Darthpbal, 2017-06-15:
3 UV 1 standalone 2 lum combo

bmpina, 2017-06-18:
ready for integration