Darthpbal/eclipse-ASUPayload #13: light to frequency

Darthpbal, 2017-05-17:


Darthpbal, 2017-05-19:
product [page](https://www.sparkfun.com/products/9768)

Darthpbal, 2017-05-20:
Performed a test including this sensor. Here is the test [results](https://docs.google.com/spreadsheets/d/1VVuWeKmqrcR2FtbpIA9LeYNiJYDSfmRQoQhWKUKncOs/edit#gid=65263359).

The light to frequency converter was tested while implemented using interrupts to count the signal frequency. 

This test was the first indication that the light to frequency converter can cause some lag inthe log time in high brightness conditions due to the high frequency output of the sensor. The lag is indicated by the change in time between logging on the bottom graph. I believe the lag is mainly a side effect of combining a high frequency interrupt signal with one of the lower end controller clock speeds. I think that in the high brightness conditions, the high frequency interrupt is not allowing a significant number of CPU instructions to occur between interrupts, so normal operations are slowed down. 

This however wasn't  testing to see how bad the lag can be, and was only tested alongside other sensors. Though, if I'm correct about the lag being due to the combination of high interrupt frequency and relatively low clock speed, I should be able to show this delay get at least a little worse while testing this sensor alone on an arduino uno.

bmpina, 2017-07-03:
Might not need this.