Darthpbal/eclipse-ASUPayload #19: Humidity DHT22

Darthpbal, 2017-05-17:


Darthpbal, 2017-05-19:
datasheet [page](https://cdn-shop.adafruit.com/datasheets/DHT22.pdf)

hookup guide [page](https://learn.adafruit.com/dht)

bmpina, 2017-06-04:
I tested 2 of these sensors. 
#1 seems to be working fine.
#2 is giving off slightly skewed values and I believe that is due to a bad wire connection. I need a soldering Iron to fix the wiring.

bmpina, 2017-06-05:
Tested the LM35 Temp sensor we found in the lab. It works so it can be used as a backup

bmpina, 2017-06-05:
Tested the DS18B20 Temp sensor we found in the lab. It works so it can be used as a backup.

bmpina, 2017-06-05:
The LM35 temp sensor we found in the lab is an outdated sensor that does not work well for arduinos.

bmpina, 2017-06-06:
Tested the HIH-4030 humidity sensor we found in the lab. The values are a bit high and it could be due to the fact that the sensor is very sensitive to damage/ESD and it was stored improperly. I recommend we stick to the DHT22 for humidity measurements.

bmpina, 2017-07-03:
integrated into system

closed 2017-07-03 by bmpina